eoprede/prusa_profiles #9: MK3 Speed Profile PLA inhibits supports?

marhar, 2018-11-04:
Very strange thing.  As you can see from the attached screenshots, Setting Printer to "MK3 Speed Profile PLA" disables supports.  All other config parameters are the same.

Any hints as to what might be awry?
[image]
[image]

Dutchy-79, 2018-11-04:
Open print settings and turn support on.

marhar, 2018-11-04:
> Open print settings and turn support on.

Yep, that's on in both cases.

Dutchy-79, 2018-11-05:
So "Generate support material" and "Auto generated supports" are on and the "Overhang threshold" as at 45 or more?

marhar, 2018-11-06:
> So "Generate support material" and "Auto generated supports" are on and the "Overhang threshold" as at 45 or more?

Correct.  It works as expected with the stock printer profile "Original Prusa i3 MK3".  It seems there's more parameters specified in the config files than are in the GUI.

Current workaround is to use a modified stock profile, with pre and post gcode copied from Fast.  This is working well enough so no urgency, but it does seem odd that there's something in the config that stops support generation.

Dutchy-79, 2018-11-06:
Maybe the profile is meant for a different version.

closed 2018-12-28 by eoprede