eoprede/prusa_profiles #13: Unable to load config file

PCFrog, 2019-03-13:
I receive this error.

[image]

trevjonez, 2019-03-13:
what version of slic3r wrote the ini file and what version is trying to import it? they have been changing a ton of things perhaps they changed some ini file parsing? if so it might be worth reporting to the main repo that it is failing to import configs.

eoprede, 2019-03-14:
Did you see this note on the front page of the repo: 
Since this seems to be the common issue: please make sure that you either download archive of all the files, make git clone or download a raw file. Attempting to copy-paste the file or downloading a file from regular github view (non-raw) will result in errors while importing.
That's the most common reason for having issues.

PCFrog, 2019-03-14:
> Did you see this note on the front page of the repo:
> Since this seems to be the common issue: please make sure that you either download archive of all the files, make git clone or download a raw [...]

Feeling sheepish...  That one statement was just out of my browser window view.  After following that it worked. Thanks for pointing it out.

closed 2019-08-23 by eoprede

ochm, 2019-08-23:
The same error while trying to import Pretty PETG v3 to PrusaSlicer 2.0.0
Failed loading the preset file: C:\Users\PC\Desktop\PrettyPETGV3.ini
Reason: <unspecified file>(1): '=' character not found in line
The content of ini file is created by copy/paste from raw data on github.